UbiquityRobotics/learn #4: In logitech controller section asks to ping robot before you've connected to it

davecrawley, 2017-11-22:
In the logitech controller section there is a troubleshooting guide that doesn't make much sense as it asks you to ping to the robot before you've connected to it. 

We need to at least refer to how to connect to the robot before telling people to do something like that.

jim-v, 2017-12-09:
It seems reasonable to use `ping` to check that the robot is up, and its name is resolvable before connecting to it.  Maybe the documentation should explain the purpose of each command?

jrlandau, 2018-01-22:
@davecrawley  I think the problem here is that there are 2 degrees of connection here--pinging, and talking to the OS.  I believe it does make sense to ping the robot before you can do the latter.

MoffKalast, 2022-05-18:
This has been addressed at some point, and [Using the Optional Logitech Controller](https://learn.ubiquityrobotics.com/noetic_quick_keyboard_driving#using-the-optional-logitech-controller) is now a separate linkable section. Though we should probably merge the Troubleshooting part into either [Tips and Tricks](https://learn.ubiquityrobotics.com/noetic_quick_start_tips_and_tricks) or [Diagnostics & Troubleshooting](https://learn.ubiquityrobotics.com/noetic_magni_silver_diagnostics_and_troubleshooting) and link to that, so we don't have the same writeup in 2 different places.

closed 2022-05-18 by MoffKalast