UbiquityRobotics/learn #10: Bad advice regarding hosts file

jim-v, 2017-12-03:
The documentation tells the user to append lines to the hosts file.
https://github.com/UbiquityRobotics/learn/edit/master/doing_more/workstation_setup.md#L64

The purpose of the hosts file is to provide name to IP address resolution in both directions.  It is necessary to have an entry for the host's own name so it can connect to itself.  However, when the name associated with the host's IP address is looked up, the first entry that matches will be returned, which, will contain the old name if these instructions are followed.  Therefore it is better to edit the hosts file and *replace* the old entry.  Also note that the addresses associated with the loopback interface are in the 127 network, not then 128.

Also, it is considered good practice to have a single entry for each IP address, containing the official hostname and aliases, e.g.:

```
127.0.1.1	yoyodyne yoyodyne.local
```

Type `man 5 hosts` for more information.

rohbotics, 2017-12-20:
The hostname change should be handled with pifi now as of version 0.4.0.

Simply use `sudo pifi set-hostname NEW_HOSTNAME`, this should greatly simplify the documentation.

jim-v, 2017-12-20:
I tested pifi 0.4.0 on a magni running the 2017-11-12 image, and changing the hostname worked as expected.  After that, I was able to view images, echo topics, etc. on my laptop, after I updated `ROS_MASTER_URI` to point to the new name.

rohbotics, 2017-12-23:
@anfederman @jrlandau Can you update the documentation to use pifi to set the hostname?

jrlandau, 2017-12-24:
done.  Please check that I've got it right.  It's in workstation setup.

rohbotics, 2017-12-24:
@jrlandau I think that setting the hostname should be earlier, pehaps during the network connection step, that way you can have multiple robots on your network and be able to address them all.

jrlandau, 2017-12-26:
done.

closed 2017-12-26 by rohbotics