UbiquityRobotics/learn #11: Printing markers

anfederman, 2017-12-06:
I added a fiducial follow (409) marker into doing_more  and added a link from the fiducial follow page. There are instructions for printing markers in the fiducials docs, but the script

rosrun aruco_detect create_markers.py  etc didn't work for me, because I had the wrong syntax, but got no error message explaining the correct syntax.

cd to the scripts directory and

python  create_markers.py did 

my mistake is I didn't specify "409 409 409.pdf"   I missed one '409'

rohbotics, 2017-12-08:
Is this an issue in the learn documentation? Is it fixed?

Whoever does fix it, can you please close this issue when you do?

Thanks

jim-v, 2017-12-08:
`rosrun aruco_detect create_markers.py` was not working because the script was not installed in the correct directory.  This has been fixed in aruco_detect 0.7.5, which will be available after the next sync.

jim-v, 2017-12-13:
It works for me with the latest:

```
rosrun aruco_detect create_markers.py 
Usage: /opt/ros/kinetic/share/aruco_detect/create_markers.py startId endId pdfFile [dictionary]
```

rohbotics, 2017-12-16:
@anfederman 

Can you test this again with the latest `sudo apt update; sudo apt upgrade` ?

rohbotics, 2017-12-27:
Closing because issue seems to be resolved in the latest version.

closed 2017-12-27 by rohbotics