UbiquityRobotics/learn #12: wget not qorking for binary files.

anfederman, 2017-12-23:
I tried to use wget for the loki catkin-ws/src files,  but they were corrupted.  I used ftp instead.  I am not sure if downloading them and then ftp will work, I am not sure how the files are getting messed up.  The error I got was the file was not in gz format.  wget is binary by default, I thought, and these are pretty small files.

rohbotics, 2017-12-23:
What was the exact wget command you ran?

rohbotics, 2017-12-24:
Can you confirm that `wget https://raw.githubusercontent.com/UbiquityRobotics/learn/master/quick_start/loki/bus_server.tar.gz` works.

rohbotics, 2018-01-12:
@anfederman Can you test so we can close this issue?

anfederman, 2018-01-13:
I have tested on my ubuntu laptop.  gunzip + tar -xf  were able to unpack.  For some reason on some systems tar -xfz does not work.  Not sure why this is.

closed 2018-01-13 by anfederman