UbiquityRobotics/learn #13: Documentation for Using our image without a robot

rohbotics, 2018-01-12:
We are getting quite a few downloads of our image, we need to make sure that people are happy with their experience. 

By the February Image (targeted for 2/18), I want to be able to point people with magni's loki's and just pi's to our image and have a reasonable expirence. 

Things that need to be done for this:

- [ ] Create a section/page for the robot-less mode.
- [ ] Have some instructions for keyboard/video/mouse folks
- [ ] Document some of the quirks of the image pi-miniuart-bt, RTC chip, etc
- [ ] If auto-detect doesn't make it, explain how to disable the magni service

rohbotics, 2018-01-22:
Not exactly, the provisioning setup will be pretty different. 
And what you see when you plug in a keyboard/video/mouse is separate.

Anything tutorial-like greater than that is out of scope. We just need the minimum for people to get up and running, and then they are on their own.

rohbotics, 2020-03-06:
So turns out this page exists now. https://learn.ubiquityrobotics.com/image_no_magni

There are still some things missing from it.

mjstn, 2020-03-06:
I believe this is all done as of this date when disable of roscore was also added.
We can add things here as forum inputs tell us other things are required.

We do have a FOLLOWUP ISSUE that I would like to see done on our image distribution page!

I would like to see a statement about users of our image that do NOT have a Magni should be directly pointed to this specific page.   Nobody would follow up and browse our overall learn pages or maybe don't even know about them from the image download page.

Doing this will i am sure greatly lower the forum questions from non customers.

jrlandau, 2020-03-08:
Should this issue be closed, or is there still something to do?

mjstn, 2020-03-08:
I think all the key big points have been hit and our goal is to minimize questions from non-customers bogging down the forum.  Now we just point them to this for 95% of what is likely to be asked is on this now.

I say close this but let Rohan have final word on if it can be closed.

This page is meant to minimize questions that are asked over and over that are making support of the forum time consuming for non Magni customer questions.  We like them but must make it clear we try to help but are not here to hold their hand for their projects.

So I feel we consider this done as it has a huge head start and then as we see things get asked that we consider important enough to add to this page OR that get asked several times we info here simply to minimize forum future post time effort.

jrlandau, 2020-03-08:
Rohan, what do you think?

rohbotics, 2020-03-08:
A couple more things I would like in a section about the GUI: "The GUI is provided for debugging purposes only, the primary supported method of using this image is headless via SSH. Minor GUI issues probably exist, but we don't have the resources to address them."  "If you are trying to connect to Wifi from the GUI you must first 'disconnect' from the `ubiquityrobotXXXX` network"

Also a note that we have `pi3-miniuart-bt` enabled by default, and so bluetooth stability may be affected, disable it if you are using bluetooth but not the serial port.

jrlandau, 2020-03-08:
I added these to the image page (#2 under quickstart).

rohbotics, 2020-03-08:
Looks good to me. Thanks!

closed 2020-03-08 by rohbotics