UbiquityRobotics/learn #22: Depricated command in documentation for Magni_nav

davecrawley, 2018-05-02:
In the documentation here:
https://ubiquityrobotics.github.io/learn/rviz

It says:

On Magni, you can type:

roslaunch magni_viz view_nav.launch when navigation is running
or
roslaunch magni_viz view_robot.launch when it is not.

This doesn't work - I believe the issue is that this has become deprecated I believe the current correct command is:

roslaunch magni_nav move_base.launch

Please fix

jim-v, 2018-05-03:
The launch files in `magni_viz` exist in the source repository, but a release has not been made since they were added.  I thought we had come up with a scheme for having documentation that was accurate for both released and unreleased versions of the software.

davecrawley, 2018-05-04:
I think we did a patch for the problem (regress the documentation to a
point in time where it was believed everything was in sync). However this
is not a flawless process, and evidently didn't work this time.

It seems that the correct solution is to do a release.

David

rohbotics, 2018-07-22:
Closed as magni_viz is released now

closed 2018-07-22 by rohbotics