UbiquityRobotics/learn #38: Unclear on how to get specific version of firmware

rohbotics, 2020-02-15:


JanezCim, 2020-08-02:
Firmware updating tutorial https://learn.ubiquityrobotics.com/firmware-upgrade
1. "If a newer version is available" - "If an older firmware version is needed,"
2. This sentence i needed to read a couple of times to understand it. "Sometimes improvements to our motor controller firmware, in this case v35 or later, make it necessary to first upgrade the linux host side software before installing the newest firmware. Our host side software supports prior firmware versions. To update host software use a process we call ‘Linux Host Software Update’ seen on this page."  I would make it simpler, example: "V35 and newer need the latest linux host side software. Instructions for updatng the linux host side software can be found HERE(link)."
3. "Enter this token and either accept the default version which is seen in the above table or specify a different version perhaps for latest features." - i would give an example that you have to insert "v35" and not just "35"

Verification tutorial https://learn.ubiquityrobotics.com/verification
1. "In the teleop window press the i letter key at a quick rate for 4 seconds. This should move the Magni about 0.6 meters forward which i about one spin of each wheel since one rotation is very near 0.64 meters." - "which **i** about one spin?" Also in 4 secs it should do 1.2m, not 0.6m
2. "In the teleop window press the i key repeatedly at a fast rate (5 times a second) and the wheels spin in a forward direction." - You can just hold the i key and it works the same
3. "In the teleop window press the k key once to be in “stop” mode then press the z key several times until the “speed” value shows a value close to 0.32 meters per second. If you go too far the , (comma) key backs the speed value down." - on english keyboard the keys are w and x to speed up or down the teleop. "Comma" key might be incorrect instruction here

rohbotics, 2020-08-02:
@JanezCim This is great feedback, thanks.

closed 2020-08-04 by JanezCim