UbiquityRobotics/learn #53: ur50 page may need tweek on IP number shown and the picture of magni with a lidar

mjstn, 2022-03-22:
Issues with ur50_lidar_setup_common page on learn. Nice to see this happening. I want to know proper repo to log an issue. Here is summary:

-  The default location of lidar says ‘laser’ on the lidar. This will confuse people so my ‘nitpik’ issue here is make that image but say ‘Lidar’ on the darn ‘lidar’.
- I suspect there are some numbers wrong in point 2 of ‘Advanced: Changing Lidar internal IP settings’. It says ‘By default the PC IP should be 192.168.1.125’ I feel this is likely wrong. Why would it not be something like [192.168.42.xxx](http://192.168.42.xxx/) ?? Please verify that.

JanezCim, 2022-03-22:
> The default location of lidar says ‘laser’ on the lidar. 

Yeah "laser" is actually default frame name for the TF frame of the lidars, so that is why that name is there. I think im going to take the second advice and under the pic write "Lidar is the component that says 'laser' on the image (lidars are sometimes also called lasers)"

>  suspect there are some numbers wrong in point 2 of...

agreed, this was explaned a bit porly. I will change it with:

`in the video PC IP  is changed from 192.168.1.125 to 192.168.42.125. This was done because often external network is also on the subnet 192.168.1.xxx and this creates problems with the internet connection. `

JanezCim, 2022-03-22:
Check and reopen if necesseary

closed 2022-03-22 by JanezCim