UbiquityRobotics/learn #59: Where would be the best place to have Conveyorbot images saved?

JanezCim, 2022-05-20:
We need to post this image https://ubiquity-pi-image.sfo2.digitaloceanspaces.com/2022-05-16-breadcrumb-noetic-improve-move_smooth.img.xz

somewhere where customers could find it. At the same time, this page should serve as a repository of all the future images that we release for conveyorbot. 

should it be: 
 - with the image downloads https://learn.ubiquityrobotics.com/noetic_pi_image_downloads with the other images
 - under exsisting conveyorbot links, eg. introduction?
 - under new conveyorbot page which is made for images? 
 - other?

MoffKalast, 2022-05-20:
Hmm yeah I think it would make most sense to make another page under conveyorbot specifically for these image downloads. Are you guys sure this ought to be public though? I was under impression that people needed to buy it first.

JanezCim, 2022-05-23:
yes, you're right, talked with david, and bc will not be public.

closed 2022-05-23 by JanezCim