UbiquityRobotics/learn #66: Clearly labelling what's supposed to be run on the Workstation and what on the Robot

MoffKalast, 2022-08-17:
For example, the https://learn.ubiquityrobotics.com/noetic_quick_start_ros101 page isn't clear on which parts need to be installed where, which lead to people attempting and failing to install ros-desktop-full on the robot.

anfederman, 2022-08-17:
I forget which version of unbuntu ARM32/64, but one of them doesn't have
desktop-full, so you need separate installs of things like that and other
components.

MoffKalast, 2022-08-17:
Yep, the 32 bit one. And that's not really the reason, but more that the Pi can't possibly run Gazebo or Rviz at any decent speed.

anfederman, 2022-08-17:
If I recall, I did a desktop - full on my Noetic ARM64 build, but there
were some things that I needed to apt install like image-compression and
velocity controllers that were missing. Also y needed to get most of the
Ubiquity stuff from gift, and then fix missing dependencies as they came
up. I think these are in my build notes in the git issues on pi2 image.