UbiquityRobotics/learn #74: Improving the Conveyorbot docs

MoffKalast, 2023-02-16:
I've ported over the recent changes from breadcrumb_learn and split the merged content onto these pages:

- Introduction (basic product description)
- Basic Usage (for people just using the robot)
- Fiducial Markers (general info about markers and also how to generate them)
- Route Setup (how to set up routes from start to finish)
- Advanced Documentation (anything that requires terminal access to do)

They can all be found under https://github.com/UbiquityRobotics/learn/tree/master/noetic/conveyorbot

Here's a list of things that need to be addressed in the conveyorbot section of the learn docs, based on my observations and old issues in the now deprecated breadcrumb_learn repo in no particular order:

- [ ] An extended explanation about setting up the routes in a clockwise direction
- [ ] Update old aruco images with new stag ones on all pages
- [ ] Fix the route layout demo images so they don't loop in the wrong direction, change arucos to stags on them
- [ ] The basic usage page could probably use more operating info, like keeping robots charged and what to do if the robot is stuck, etc., just things that people using an already set up robot would need to know
- [ ] The touchscreen section of the route setup page needs to be completely rewritten so it features an image for each step of the setup, walls of text need to be broken into shorter paragraphs, likely also needs to be brought up to speed with recent changes, see code of [ezmap getting started guide](https://learn.ubiquityrobotics.com/noetic_ezmap_gettingstarted) for markdown examples of embedding images in text for greater specificity 
- [ ] Note the weight limits of the tower shelves somewhere
- [ ] Downloadable marker sets for testing (has its own issue if it ends up being a separate site, but we should note it here as well, since it needs to be linked to or embedded on the learn page)
- [ ] A troubleshooting guide with frequent issues and how to solve them

anfederman, 2023-02-16:
I could only help if I had a convoyorbot.  Is the one still being tested
for radio interference available?

MoffKalast, 2023-02-16:
Hey Alan, as of right now I'm not convinced the existing doc is enough to really get to know the full ins and outs of the system yet, so you'd probably have some trouble setting it up. 

I think Matjaž should have a good look at it first and bring it up to date with the current state, and then you can go over it and fill in what you think ought to be there from a fresh viewpoint. It would definitely benefit from a native speaker rewording a lot of it regardless.

Shipping you a unit may take some time, so we might as well get on it now. You ought to talk to @NejaE about the logistics.

anfederman, 2023-02-16:
Maybe just getting a pi 4 and working on EZmap makes more sense.