UbiquityRobotics/learn #75: Support for ROS2 Humble

rohitpandey1234, 2023-03-09:
We are interested in integrating ubiquity lidars in our robots, may you please clarify is it possible to get ROS2 support

MoffKalast, 2023-03-09:
Well we are slowly working towards ROS 2 support, but it is unlikely to be available all that soon. For now I'd suggest installing both ROS 1 and 2, setting up the [ros1_bridge](https://github.com/ros2/ros1_bridge) and then running the lidar nodes under ROS 1, with the rest of your system on the ROS 2 side.

anfederman, 2023-03-09:
I saw a ROS2 driver for a LDLIDAR, which might work for our 5 meter lidar
which I think is an LD clone. Not sure if the Lesein Lidar has anything
similar for ROS2, though some manufacturers like Ouster have nodes.

anfederman, 2023-03-09:
https://github.com/Myzhar/ldrobot-lidar-ros2   the cmd_interface_linux.cpp
looks like cut and paste from our ur12_lidar code.

anfederman, 2023-03-09:
Also found this: https://github.com/LS-Technical-Supporter/LS-LIDAR-C32-ROS2

our ur50_lidar is sourced from this company.

MoffKalast, 2023-03-09:
Well that looks promising, we just need someone to verify that they work.

anfederman, 2023-03-10:
Well, get me a darn RPi 4 and the conveyor bot sitting in los Gatos and I
can start testing.rhw big deal will be Ubiquity motor in ROS 2.