UbiquityRobotics/learn #77: Ubiquity apt key download appears broken

erelson, 2023-12-17:
Decided to update an old install, and pull packages. Found the apt setup instructions, but ran into the below when trying to setup the needed key.

https://learn.ubiquityrobotics.com/kinetic_package_repositories

```
$ sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key C3032ED8

Executing: /tmp/tmp.rfWnAbQvgH/gpg.1.sh --keyserver
hkp://ha.pool.sks-keyservers.net:80
--recv-key
C3032ED8
gpg: requesting key C3032ED8 from hkp server ha.pool.sks-keyservers.net
?: ha.pool.sks-keyservers.net: Host not found
gpgkeys: HTTP fetch error 7: couldn't connect: Connection timed out
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: keyserver unreachable
gpg: keyserver communications error: public key not found
gpg: keyserver receive failed: public key not found
```

Looks like this instruction was added here:
https://github.com/UbiquityRobotics/learn/commit/ead08afa3fb48a45fea3fd4334cd9a3056d1428e#diff-53780ad7c84ae253521d5d069fb6100c3e2a0499ea4c02abe8720c3bc743419eR24

Another report of this issue: https://forum.ubiquityrobotics.com/t/i-cant-download-ubiquity-robotics-keys/1039

My particular setup is:

```
$ lsb_release -a

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.7 LTS
Release:        16.04
Codename:       xenial

$ uname -r

4.14.98-v7+
```

Looks like I already have a valid key on this install, though, so my apt installs work fine.

anfederman, 2023-12-17:
> ead08af
> #diff-53780ad7c84ae253521d5d069fb6100c3e2a0499ea4c02abe8720c3bc743419eR24

You are correct the apt  key is broken. You can download all the Ubiquity
code into a ros_ws from our GitHub. This even works for ARM 64 Noetic on
raspberry pi 4.

anfederman, 2023-12-17:
Also our current code is Ubuntu 20 ROS Noetic. Ubuntu 16 isn't supported
any more.

closed 2025-06-08 by erelson